Berny23/LD-ToyPad-Emulator #236: sudo apt update on vmware terminal not working

Ausbruhh, 2025-04-03:
i have followed all steps correctly and everything has gone smoothly up until the point when i have to copy and paste the first lines of code from the README into the terminal on the VM and i end up with a bunch of errors, i have bridged the connection thinking that was the issue and ive also tried messing with some of the preferences with the same outcome, i rlly wanna play this game so help will be much appreciated
my dc is ausbruhh for a quick response

[image] start of command

[image] end of command

Florin9doi, 2025-04-05:
your vm isn't connected to internet

Ausbruhh, 2025-04-07:
> your vm isn't connected to internet

really? do you know how to because i thought thats what the 'bridging' thing is

Ausbruhh, 2025-04-10:
ive tried all connection methods listed here, please someone help 

[image]

14thLetter, 2025-07-02:
Have you gotten any solutions, as I have this SAME exact issue