Berny23/LD-ToyPad-Emulator #248: Raspberry Pi Zero 2w error when attempting to start

Zukariin, 2025-05-12:
i followed the instructions for the Pi and every time i get to npm run start i get these errors

gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/lego/.nvm/versions/node/v11.15.0/lib/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
gyp ERR! System Linux 6.12.25+rpt-rpi-v7
gyp ERR! command "/home/lego/.nvm/versions/node/v11.15.0/bin/node" "/home/lego/.nvm/versions/node/v11.15.0/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/lego/LD-ToyPad-Emulator/node_modules/bufferutil
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v8.4.1
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @rollup/rollup-linux-x64-gnu@4.9.5 (node_modules/@rollup/rollup-linux-x64-gnu):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for @rollup/rollup-linux-x64-gnu@4.9.5: wanted {"os":"linux","arch":"x64"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bufferutil@3.0.0 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bufferutil@3.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/lego/.npm/_logs/2025-05-12T15_02_27_480Z-debug.log

JonasBrinkmann-DE, 2025-05-12:
Do you see any errors in the console, was the toypad detected by the game?

JonasBrinkmann-DE, 2025-05-12:
@Zukariin I found the bug in the code, turns out I broke the main branch once again(at least for tag deletion). I opened up a pull request for the fix. Until then replace the index.js file with the one from my pull request: https://github.com/JonasBrinkmann-DE/LD-ToyPad-Emulator/blob/patch-1/index.js

Zukariin, 2025-05-12:
> Do you see any errors in the console, was the toypad detected by the game?

Initialized toytags.JSON
<<Tags are synced!>>
#Vehicle Deletion Not Processing
IO Recieved: Deleting entry 0417771430A280 from JSON
Entry to delete:  0
Token was not found.
#Character creation erroring
Creating character: 1
Error: ENOENT: no such file or directory, open '/home/lego/LD-ToyPad-Emulator/server/json/charactersmap.json'
    at Object.openSync (fs.js:448:3)
    at Object.readFileSync (fs.js:348:35)
    at getCharacterNameFromID (/home/lego/LD-ToyPad-Emulator/index.js:81:19)
    at app.post (/home/lego/LD-ToyPad-Emulator/index.js:503:16)
    at Layer.handle [as handle_request] (/home/lego/LD-ToyPad-Emulator/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/lego/LD-ToyPad-Emulator/node_modules/express/lib/router/route.js:149:13)
    at Route.dispatch (/home/lego/LD-ToyPad-Emulator/node_modules/express/lib/router/route.js:119:3)
    at Layer.handle [as handle_request] (/home/lego/LD-ToyPad-Emulator/node_modules/express/lib/route

i went through all the files and they appear to have downloaded correctly so i have no idea and i tried resetting it up and same problems.

JonasBrinkmann-DE, 2025-05-12:
Just pushed another pull request. In index.js line 24 or so where it says __dirname, "server/json/charactersmap.json" change charactersmap to charactermap

JonasBrinkmann-DE, 2025-05-12:
[image]

JonasBrinkmann-DE, 2025-05-12:
That should fix it