Berny23/LD-ToyPad-Emulator #253: step 6 is giving me an error!!

mustafael05, 2025-05-13:
Hello, I get a long error message in the terminal when I paste and run the code from step 6 of the tutorial. 

[image]
[image]
[image]
[image]

JonasBrinkmann-DE, 2025-05-14:
Try running: npm install bufferutil@3.0.0
And then rerun the commands.

mustafael05, 2025-05-14:
now I got this error

[image]
[image]
[image]

glitchygeorge, 2025-05-31:
this is an issue for me as well idk whats causing it