Berny23/LD-ToyPad-Emulator #259: I need help...

Felipe20Rich, 2025-05-28:
I followed the video tutorial at the link: https://www.youtube.com/watch?v=5PARAnrt1jU correctly and it gives me these errors when I include the third command that is requested...

(Debian 11, Linux) step 6: 

curl -o- https://raw.githubusercontent.com/nvm-sh/nvm/v0.40.3/install.sh | bash
export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && \. "$NVM_DIR/nvm.sh"
[ -s "$NVM_DIR/bash_completion" ] && \. "$NVM_DIR/bash_completion"

nvm install 11
sudo setcap cap_net_bind_service=+ep `readlink -f \`which node\``
npm install --global node-gyp@8.4.1
npm config set node_gyp $(npm prefix -g)/lib/node_modules/node-gyp/bin/node-gyp.js

cd LD-ToyPad-Emulator
npm install

(Output, lines final) Error:

gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/debian/.nvm/versions/node/v11.15.0/lib/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
gyp ERR! System Linux 5.10.0-35-amd64
gyp ERR! command "/home/debian/.nvm/versions/node/v11.15.0/bin/node" "/home/debian/.nvm/versions/node/v11.15.0/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/debian/LD-ToyPad-Emulator/node_modules/bufferutil
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v8.4.1
gyp ERR! not ok 
npm WARN ws@8.17.1 requires a peer of bufferutil@^4.0.1 but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bufferutil@3.0.0 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bufferutil@3.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/debian/.npm/_logs/2025-05-28T17_21_32_138Z-debug.log

Tivertoni, 2025-05-28:
Try running these commands and then try again:
sudo apt update
sudo apt install -y build-essential python3 make g++ gcc

Tivertoni, 2025-05-28:
Did you type the commands shown in the video or from the github page?

Felipe20Rich, 2025-05-28:
> sudo apt update
> sudo apt install -y build-essential python3 make g++ gcc

Still gives the same error...

Felipe20Rich, 2025-05-28:
> Você digitou os comandos mostrados no vídeo ou na página do github?

I just copied and pasted it, following the step-by-step instructions...

Tivertoni, 2025-05-28:
Try these, this will update some stuff: 
nvm install 20
nvm use 20
npm install --global node-gyp
cd ~/LD-ToyPad-Emulator
rm -rf node_modules package-lock.json
npm install

Felipe20Rich, 2025-05-28:
d

> Tente estes, isso atualizará algumas coisas: nvm install 22 nvm use 22 npm install --global node-gyp cd ~/LD-ToyPad-Emulator rm -rf node_modules package-lock.json npm install

npm error   919 |         length_ = string->WriteUtf8(str_, static_cast<int>(len), 0, flags);
npm error       |                                     ^~~~
npm error       |                                     |
npm error       |                                     char*
npm error In file included from /home/debian/.cache/node-gyp/22.16.0/include/node/v8-object.h:11,
npm error                  from /home/debian/.cache/node-gyp/22.16.0/include/node/v8-array-buffer.h:13,
npm error                  from /home/debian/.cache/node-gyp/22.16.0/include/node/v8.h:24,
npm error                  from /home/debian/.cache/node-gyp/22.16.0/include/node/node.h:74,
npm error                  from ../../nan/nan.h:47,
npm error                  from ../src/bufferutil.cc:7:
npm error /home/debian/.cache/node-gyp/22.16.0/include/node/v8-primitive.h:203:26: note:   initializing argument 1 of ‘int v8::String::WriteUtf8(v8::Isolate*, char*, int, int*, int) const’
npm error   203 |   int WriteUtf8(Isolate* isolate, char* buffer, int length = -1,
npm error       |                 ~~~~~~~~~^~~~~~~
npm error In file included from ../src/bufferutil.cc:7:
npm error ../../nan/nan.h: In member function ‘v8::Local<v8::Value> Nan::Callback::Call_(v8::Isolate*, v8::Local<v8::Object>, int, v8::Local<v8::Value>*) const’:
npm error ../../nan/nan.h:1477:5: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
npm error  1477 |     ));
npm error       |     ^
npm error In file included from ../../nan/nan.h:47,
npm error                  from ../src/bufferutil.cc:7:
npm error /home/debian/.cache/node-gyp/22.16.0/include/node/node.h:200:50: note: declared here
npm error   200 |                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
npm error       |                                                  ^~~~~~~~~~~~
npm error /home/debian/.cache/node-gyp/22.16.0/include/node/node.h:117:42: note: in definition of macro ‘NODE_DEPRECATED’
npm error   117 |     __attribute__((deprecated(message))) declarator
npm error       |                                          ^~~~~~~~~~
npm error In file included from ../src/bufferutil.cc:7:
npm error ../../nan/nan.h: In member function ‘void Nan::AsyncWorker::SaveToPersistent(const char*, const v8::Local<v8::Value>&)’:
npm error ../../nan/nan.h:1531:64: error: no matching function for call to ‘v8::Object::Set(v8::Local<v8::String>, const v8::Local<v8::Value>&)’
npm error  1531 |     New(persistentHandle)->Set(New(key).ToLocalChecked(), value);
npm error       |                                                                ^
npm error In file included from /home/debian/.cache/node-gyp/22.16.0/include/node/v8-array-buffer.h:13,
npm error                  from /home/debian/.cache/node-gyp/22.16.0/include/node/v8.h:24,
npm error                  from /home/debian/.cache/node-gyp/22.16.0/include/node/node.h:74,
npm error                  from ../../nan/nan.h:47,
npm error                  from ../src/bufferutil.cc:7:
npm error /home/debian/.cache/node-gyp/22.16.0/include/node/v8-object.h:240:37: note: candidate: ‘v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)’
npm error   240 |   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
npm error       |                                     ^~~
npm error /home/debian/.cache/node-gyp/22.16.0/include/node/v8-object.h:240:37: note:   candidate expects 3 arguments, 2 provided
npm error /home/debian/.cache/node-gyp/22.16.0/include/node/v8-object.h:243:37: note: candidate: ‘v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, uint32_t, v8::Local<v8::Value>)’
npm error   243 |   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
npm error       |                                     ^~~
npm error /home/debian/.cache/node-gyp/22.16.0/include/node/v8-object.h:243:37: note:   candidate expects 3 arguments, 2 provided
npm error In file included from ../src/bufferutil.cc:7:
npm error ../../nan/nan.h: In member function ‘void Nan::AsyncWorker::SaveToPersistent(const v8::Local<v8::String>&, const v8::Local<v8::Value>&)’:
npm error ../../nan/nan.h:1537:42: error: no matching function for call to ‘v8::Object::Set(const v8::Local<v8::String>&, const v8::Local<v8::Value>&)’
npm error  1537 |     New(persistentHandle)->Set(key, value);
npm error       |                                          ^
npm error In file included from /home/debian/.cache/node-gyp/22.16.0/include/node/v8-array-buffer.h:13,
npm error                  from /home/debian/.cache/node-gyp/22.16.0/include/node/v8.h:24,
npm error                  from /home/debian/.cache/node-gyp/22.16.0/include/node/node.h:74,
npm error                  from ../../nan/nan.h:47,
npm error                  from ../src/bufferutil.cc:7:
npm error /home/debian/.cache/node-gyp/22.16.0/include/node/v8-object.h:240:37: note: candidate: ‘v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)’
npm error   240 |   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
npm error       |                                     ^~~
[... cut ...]
npm error gyp ERR! command "/home/debian/.nvm/versions/node/v22.16.0/bin/node" "/home/debian/.nvm/versions/node/v22.16.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm error gyp ERR! cwd /home/debian/LD-ToyPad-Emulator/node_modules/bufferutil
npm error gyp ERR! node -v v22.16.0
npm error gyp ERR! node-gyp -v v11.0.0
npm error gyp ERR! not ok
npm error A complete log of this run can be found in: /home/debian/.npm/_logs/2025-05-28T19_02_25_343Z-debug-0.log

only these two worked:

nvm 22 installation
nvm 22 usage
npm installation --global node-gyp

Felipe20Rich, 2025-05-30:
> Maybe update nan? npm install nan@latest

I forgot to tell you ^^', but I fixed the problem, I used Raspberry instead of Linux itself and it worked