Berny23/LD-ToyPad-Emulator #268: TypeError [ERR_INVALID_CALLBACK]: Callback must be a function

UnDGCat, 2025-08-08:
I was able to get the server up and running and connected to CEMU just fine, but the emulator's a bit too buggy in actual usage. Whenever I move a tag to a spot on the pad, I get the aforementioned error in the console. The emulator seems to still work fine at first, but whenever I create a new tag the entire ToyPad gets visually wiped, but the characters still stay ingame. Refreshing the page or clicking the sync button does not work. The only way to fix this seems to be to stop the server, wait until the game doesn't see the toypad anymore, then start it back up again, which is quite inconvenient to do every time I want to create a tag.

[image]

UnDGCat, 2025-08-08:
> Cemu has a built in toypad now, no extra accessories or VM required. I'd recommend just using that.

I would, but the lack of lighting support makes some parts of the game much harder. Plus, the interface on this one is a lot more intuitive

Tivertoni, 2025-08-10:
@UnDGCat Can I just confirm that you are using the master branch?

UnDGCat, 2025-08-11:
@Tivertoni The error no longer appears in the console, but the issue with the tags disappearing from the pad and the toytags file not getting updated with the right indices still happens.

UnDGCat, 2025-08-19:
Found that writing to the file is not the source of the issue with tags disappearing, moving this to a new issue post.

closed 2025-08-19 by UnDGCat