Berny23/LD-ToyPad-Emulator #270: Toy Tags get added twice

ccfman2004, 2025-08-10:
Is it normal when adding a toy tag it gets added twice to the Toy box? Every time I add one it shows up twice. If I delete a tag, all duplicates get deleted but then come back when I add a new tag.

Tivertoni, 2025-08-10:
@ccfman2004 That behaviour you are describing is not intended, I can try and resolve those issues, can I just confirm that you are running the master branch?

I would also try and find a solution for #271, but im not sure where that could originate from.

ccfman2004, 2025-08-10:
I followed the instructions in the Readme.md file so whatever branch is in them is what I am using.

I am in the process of starting from scratch just to make sure it wasn't something I did during setup.

Tivertoni, 2025-08-10:
Do you have any logs that you can share?

ccfman2004, 2025-08-10:
Unfortunately I have already reformatted the SD Card to try again.

Tivertoni, 2025-08-10:
That should not be causing the problems you are describing.

Basically, this project has two active branches (versions): the master branch, which is the version you are currently using and experiencing issues with, and the development branch, which includes additional features but is not considered stable yet. The development branch also has compatibility problems with microcomputers like the Pi Zero 2 you are using.

I am currently working on merging most of the code from the dev branch into the master branch.

ccfman2004, 2025-08-10:
I should mention that after following Steps 3-5 of the instructions, running "npm run start" yields an error that the START script is missing despite being in the "LD-ToyPad-Emulator" directory.

If I then follow the Update instructions, then it works so I'm not sure why it won't work before running the Update commands.

ccfman2004, 2025-08-13:
I don't know if this issue is related but sometimes when I move a toy tag to the pad the game will summon a duplicate of whatever character I am currently using even if the tag I am placing is an item/vehicle.

[LD-Emu-Logs.txt](https://github.com/user-attachments/files/21746704/LD-Emu-Logs.txt)

ccfman2004, 2025-08-19:
Here is a picture of the what it looks like when you have the duplicates.

[image]

ccfman2004, 2025-08-19:
When checking my web browser's console logs (FireFox), it shows the tags being loaded twice. Trying to use the browser Debug to see where it is loading the tags twice. I don't envy your task.

dviroz13032010-netizen, 2025-08-24:
its fixed? couse its still not work  for me

dviroz13032010-netizen, 2025-08-25:
when it will be fixed?

lmorgan42, 2025-09-10:
I am also having this issue. Specifically for me, the duplicates went away when data was written to a vehicle tag but always return when a new character is made. Additionally, when a new character is made all of the characters on the pad disappear and their squares are returned to the toy box even though the game registers them still. I am using virtual box and the only unusual element of my setup is that the install failed due to a version of bufferutil that was too old (I set things up from master by just copy pasting the entire block of commands from the guide between each restart). The bufferutil failure occurred on the final npm install command as far as I could tell so i reinstalled a newer version and re-ran npm install manually. I also had the guest additions on in virtual box.

jacobewc, 2025-10-07:
I'm having the same issues as Imorgan42, wondering if anyone has found a fix?